Rejdok/STM32-Marlin-Port #2: STM32F103 support

etet100, 2018-02-27:
Hi. Could you tell me how hard would it be to use STM32F103 (maple mini) instead of F4xx ?

Rejdok, 2018-03-01:
Hello. I think it wouldn't very difficult. You should change stm hal from F4xx series to F1xx and change timers frequency.

etet100, 2018-03-02:
Hi. Thank you for your answer. It is quite hard actually. After a few hours o trials i have 463 errors left.

Rejdok, 2018-03-02:
Hello. 463 errors uts too much. I hope you build it using sw4stm32. Just create new build, from debug, then change preprocessor definition in project properties stm32f407xx to stm32f103xB or other. And exclude from build hal drivers for stm32f407. Then change linker script in project settings, to ls for your's microcontroller. You can simply generate it using stm32cubeMX.
Sorry, but makefile dosen't work.

etet100, 2018-03-02:
It's hopeless, i'm affraid. Hundreds of errors. I give up.

Rejdok, 2018-03-02:
Ok, i add stm32f103c8t6 support. Can you change timers frequency in marlin/portable (stm32_stepper.cpp and stm32_temperature.cpp?

etet100, 2018-03-02:
Thank you very much. Unfortunatelly, i'm still experiencing a lot of issues. Some include paths are missing (fatal error: stm32f1xx.h: No such file or directory), floating point ABI is set to hard, ARM_MATH_CM4 is defined instead of ARM_MATH_CM3 (for Cortex-M3 target), ADC driver seems to be missing (no stm32f1xx_hal_adc.h), USB driver too (provided one uses USB OTG mode which is not supported by STM32F103CB). I had to change some includes (#include "stm32f4xx.h" to #include "stm32f1xx.h"). And now i have more and more errors. I have  no idea what i'm doing wrong. Are you sure you are able to compile it without errors?

Rejdok, 2018-03-03:
Yep. Sorry, my bad. I'm overlooked little bit. Try it now=)

etet100, 2018-03-04:
Thanks. Now it freezes on:

```
while(hUsbDeviceFS.dev_state!=3)
  HAL_Delay(500);
```
dev_state is always = 1

Interestingly, microcontroller also hangs when i upload
freshly created Stm32CubeMX project. Damn. I'm sure my 
maple mini board is perfectly fine.

Rejdok, 2018-03-04:
Yep. It's ok. Just connect via USB your mapple mini and PC.

Rejdok, 2018-03-04:
And mc dosht hangs. Just enable debug by serial wire in sys tab.

etet100, 2018-03-04:
It is connected to PC. And serial wire debugging is enabled. I see that USB
is not working properly:

[image]

It falls into infinite loop, every time.

Rejdok, 2018-03-04:
That's strange. It's screenshot from custom project? Did you configure usb device as virtual com port in cubeMx, and  usb clock configuration? Your pc detect new device?

etet100, 2018-03-04:
This is minimal example to show how it works. Marlin code behave exactly the same. Clock settings are copied directly from code generated by CubeMx. No new device is detected by pc. 

[image]

Rejdok, 2018-03-04:
Hm, I'm remembered. State changes, when you connect to com port using terminal. I found mc stm32 on M3, and seen, that communication doesn't work. So I'm fix it now.  Sorry for troubles. I wrote this port about year ago.
But at my mc USB COM port works. And pc found it. Try check COM port section in device manager.

Rejdok, 2018-03-04:
Ok now it works as expected.

etet100, 2018-03-05:
And still no connection. I have no idea what is going on. I have checked everything many times. Maple board is fine (it's working perfectly when i upload original maple firmware), cable are fine, firmware seems to be fine.

Rejdok, 2018-03-05:
https://os.mbed.com/media/uploads/hudakz/maple_usb.png
Try to set PB9 in high level.

etet100, 2018-03-06:
You are right! I have replaced Q2 with bridge. Of course it doesn't mean that everything is working. Communication is not stable, motors are not working, timers are not working. I found a few bugs.

Rejdok, 2018-03-08:
Timers work's now. 
Communications should be stabile.

bhgv, 2018-07-13:
hi, i send a PR with some editings and addings. mostly for maple mini. README.md and a pair of alternative ROM uploaders for people without stlink.

you may see how it looks here:
https://github.com/bhgv/STM32-Marlin-Port

etet100, 2018-07-15:
Hi. I wonder what you think about using more recent Marlin v2 which introduces hardware abstraction layer.

bhgv, 2018-07-15:
@etet100 
Give me please a link to v2

Is it your https://github.com/etet100/marlin_cetus_3d ?

Rejdok, 2018-07-15:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x . In Marlin v2 developers made a lot of great changes in code style and project organisation. But at present this branch unstable and actively develop. So they can easily change functions definitions, project structure e.t.c. That why I dont want using Marlin v2 now. In fact, if I had more time, I will port smootheware on stm32, and use this firmware. Sorry for bad english.