Rejdok/STM32-Marlin-Port #4: not an issue, but a question about of marlin g-codes

bhgv, 2018-07-19:
i'm playing with your (and etet100) stm32 ports of marlin. i'm trying to support of marlin in my [lua.gcode.sender](https://github.com/bhgv/lua.gcode.sender).

(i test a controller without connection to steppers)

when i send for example `G1X100Y100Z100F100` , and immediately after - `M114` (get current position info if i understood correctly) it give me x:100 y:100 z:100 without gradual change of coordinates. 

perhaps some preliminary settings are needed? or M114 returns not the current position of he tool?

could you give me some small, but workable on your marlin's port g-code for testing. and could you tell me with which command of marlin i can get the current position of the tool?

etet100, 2018-07-23:
Hi. Why don't you use original marlin? My fork is modified to use with specific printer (Cetus 3D). 

I'm not sure but i'm guessing that this is how M114 works. This is not real position but ending point of currently processed segment.

Rejdok, 2018-07-25:
Sorry. I didn't see your question. In current realisation M114 returns last seted position. If you want know actual position, then extract current steps from stepper, translate it to angles, then solve forward kinematics problem for your printer.

bhgv, 2018-07-25:
thank you for your responces. 
yes, i asked this question on the main marlin repo and looked in the source of marlin. it set the ending position of the segment before sending it to ..(? if to send 3 lines: G1X100 \n G1X5 \n M114 , it get X=5 immediately. no any X=100). now i  am looking the way to get the real position. i think this is a good idea (the position).

@etet100 
> Why don't you use original marlin? My fork is modified to use with specific printer (Cetus 3D).

i don't want to do a big work to porting marlin on my maple mini. your source works after minimal changes unlike the original marlin v2 (maybe later i will have more time to port it).

@Rejdok 
sadly i can't get the real position in steps. i think i should do this ability by myself. 

i think this issue(s) may be closed. thank you both.

Rejdok, 2018-07-27:
No problem, look up stepper isr code. There you can add own step counter or use exsiting.

closed 2018-07-27 by Rejdok