Rejdok/STM32-Marlin-Port #8: Pins for STM32F407zet

LordsWars, 2020-09-13:
Hello, where does the processor type switch?
I can not find where the conversion of STM pins to ARDUINO takes place ??? that is, PA1 in 10 pin arduino? I'm trying to build a marlin for the Robin 2 board where STM32F407ZET is installed, this is the first version of the compiler program that compiles at all) the rest is just not even in the database going into a file :( I work mainly with KEIL and have not yet dealt with arduino, but apparently I have to ... I assemble a 3D printer with a replaceable head (plastic printing, a milling cutter (the main task is drilling) and a laser), the printer is made for faster prototyping (PCB + housing) ....

Rejdok, 2020-10-21:
Hello. Sorry for long answer. If I remember correctly processor type can be canged in sw4stm32 interface. Pins are mapped in STM32-Marlin-Port/Marlin/Portable/port.cpp at least for 407ze. You can cange it as you like. Ported functions are also definded there.

But I recommend fork original marlin 2.0 and port to own board. It already has HAL for stm processors.

LordsWars, 2020-10-22:
I tried to compile the downloaded version of the latest marlin 2 with HAL and I can't get it to compile normally, with errors all the time. A particularly problematic place is the connection of the screen with the FMSC. What I just didn’t do the screen didn’t work. separately in keil, I launched the screen as an ILI9341 controller, but the correct positioning of the point did not occur and, accordingly, the text was not output ... then the search by ID 8066 determined that it was either RM68140 or ILI9341, after reading the documentation, I determined what the error was (in register 2A and 2B recorded 2 dates and then the next register was selected and 4 dates need to be written on the datashit) and eliminated it, which made it possible to work normally with the screen. In general, I fight like a fly on the glass and I'm already thinking that it's probably easier to write my own than trying to fit someone else's code to what is required ...

LordsWars, 2020-10-22:
Now I'm trying to figure out a touch panel ... how to work with it and which pins go to SPI1 and where the BUSI and INT pins are not clear where they go to the processor ... in general, I chop the board with a multimeter, I call it because the circuit does not correspond to the board in everything

Rejdok, 2020-10-22:
Here [example ](https://github.com/afiskon/stm32-ili9341) of using display with ILI9341. It used SPI, not 8080, but it's fine for start. How you build marlin 2.0? You built it with platformio? Multimiter isn't best choise for digital signals, it have very low update time, do you have logic analyzer? 
Marlin 2.0 using this [project ](https://github.com/danieleff/STM32GENERIC) for HAL. Checkout [this](https://danieleff.github.io/STM32GENERIC/board_BLACK_F407ZE/) page for understod pin mapping in this HAL. Actiualy you can write own pin mapping. But it doesn't desctibed well.

LordsWars, 2020-10-23:
для компиляции использую Visual Studio Code и platformio. мультиметр использую для физисеского определения рассположения контактов откуда и куда идёт провод :) а логический анализатор конечно есть и используется по назначению :) у меня специальность ремонт и эксплуатация электронных вычислительных машин :) а что касается экрана то там не просто 8080 там прямое управление от МК (FSMC) то что есть в Марлин 2 предназначено для stm32f103 а у меня процессор stm32f407 по идее должно всё подходить отличий немного и просмотрев код я добавил отличия.... у меня проблемма не с кодом а с пониманием того что нужно включать для нормальной работы экрана, не понимаю я зачем сделана такая сложная структура ведь всё можно написать намного проще...да еще и описание Марлин ну совсем не полноценные ибо описаны только 2 файла конфигурации и всё...
dlya kompilyatsii ispol'zuyu Visual Studio Code i platformio. mul'timetr ispol'zuyu dlya fiziseskogo opredeleniya rasspolozheniya kontaktov otkuda i kuda idot provod :) a logicheskiy analizator konechno yest' i ispol'zuyetsya po naznacheniyu :) u menya spetsial'nost' remont i ekspluatatsiya elektronnykh vychislitel'nykh mashin :) a chto kasayetsya ekrana to tam ne prosto 8080 tam pryamoye upravleniye ot MK (FSMC) to chto yest' v Marlin 2 prednaznacheno dlya stm32f103 a u menya protsessor stm32f407 po ideye dolzhno vso podkhodit' otlichiy nemnogo i prosmotrev kod ya dobavil otlichiya.... u menya problemma ne s kodom a s ponimaniyem togo chto nuzhno vklyuchat' dlya normal'noy raboty ekrana, ne ponimayu ya zachem sdelana takaya slozhnaya struktura ved' vso mozhno napisat' namnogo proshche...da yeshche i opisaniye Marlin nu sovsem ne polnotsennyye ibo opisany tol'ko 2 fayla konfiguratsii i vso...
Развернуть
837/5000
for compilation I use Visual Studio Code and platformio. I use a multimeter to physically determine the location of the contacts from where and where the wire goes :) and of course there is a logic analyzer and is used for its intended purpose :) I specialize in the repair and operation of electronic computers :) and as for the screen, there is not just 8080 there is direct control from MK (FSMC) what is in Marlin 2 is intended for stm32f103 and I have a stm32f407 processor, in theory, everything should fit the differences a little and after looking at the code, I added the differences .... I have problems not with the code, but with an understanding of what needs to be turned on for normal operation screen, I don't understand why such a complex structure was made, because everything can be written much easier ... and even the description of Marlin is not at all complete, because only 2 configuration files are described and that's it ...

LordsWars, 2020-10-23:
I mostly wrote simple programs, but I worked with almost all peripheral devices of MK stm32, that is, I can work separately with each device, but when collecting everything in a heap, questions and incomprehensible moments begin :) I plan to collect everything based on RTOS, but these are plans :) now I collect FSMC, DMA, SPI, I2C, CD-Card, IRQ, TIMER, USART modules in keil and try to make everything work together ... separately everything works, but when assembled in a heap, not everything works or does not work as it should. .... I'm actually more of an electronics repairman :) it takes me 30 minutes to replace the BGA144 :) it's easier to repair than to write a full-fledged program :) from here I write slowly and try to write so that the performance does not drop and everything works stably. I try not to use HAL and try not to rise higher than CMSIS because there is a lot of unnecessary things to slow down the system

Rejdok, 2020-10-23:
Когда я разрабатывал по для 3д принтеров, я наоборот пытался использовать hal и не погружаться сильно в дебри периферии, это сильно ускоряет разработку, позволяет портировать под другие платформы, а компенсировать много ненужных вещей можно просто скоростью процессора и объемом памяти. Сложные структуры как правило делаются для удобных абстракций, с которыми приятно работать, ради переносимости или просто потому что. =) 
Если не сложно, можете скинуть код, в котором конкретно проблемы возникают, не охота мне опять перерывать исходники марлина.=)

Rejdok, 2020-10-23:
Я запускал похожий дисплей по spi. Все более-менее работало. Т.к. дисплей обновляется не слишком часто, вполне можно использовать.

LordsWars, 2020-10-23:
here is the complete distribution kit, it compiles, but the screen is empty ... even the screen backlight does not turn on, I separately attach a circuit board from the MKS ROBIN2 V1.0 + TFT35V1.0 website
[MKS Robin2 V1.0_002 SCH.pdf](https://github.com/Rejdok/STM32-Marlin-Port/files/5429640/MKS.Robin2.V1.0_002.SCH.pdf)

[Marlin-bugfix-2.0.x.zip](https://github.com/Rejdok/STM32-Marlin-Port/files/5429663/Marlin-bugfix-2.0.x.zip)

LordsWars, 2020-10-23:
touch panel xr2046
and a photo of the board with the screen
[image]

LordsWars, 2020-10-23:
I program with mini STLink v2
r as soon as I turn on -DHAS_GRAPHICAL_TFT heaps of errors pour in

LordsWars, 2020-10-23:
Compiling .pio\build\MKS_ROBIN2\src\src\lcd\tft\ui_480x320.cpp.o
In file included from Marlin\src\lcd\tft\../menu/menu_item.h:24:0,
                 from Marlin\src\lcd\tft\touch.cpp:27:
Marlin\src\lcd\tft\../menu/menu.h: In static member function 'static void MenuItem_back::action(const char*)':
Marlin\src\lcd\tft\../menu/menu.h:89:63: error: 'class MarlinUI' has no member named 'go_back'
     FORCE_INLINE static void action(PGM_P const=nullptr) { ui.go_back(); }
                                                               ^~~~~~~
In file included from Marlin\src\lcd\tft\touch.cpp:27:0:
Marlin\src\lcd\tft\../menu/menu_item.h: In static member function 'static void MenuItem_submenu::action(const char*, screenFunc_t)':
Marlin\src\lcd\tft\../menu/menu_item.h:42:74: error: 'class MarlinUI' has no member named 'save_previous_screen'
     static inline void action(PGM_P const, const screenFunc_t func) { ui.save_previous_screen(); ui.goto_screen(func); }
                                                                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\tft\../menu/menu_item.h:42:101: error: 'class MarlinUI' has no member named 'goto_screen'
     static inline void action(PGM_P const, const screenFunc_t func) { ui.save_previous_screen(); ui.goto_screen(func); }
                                                                                                     ^~~~~~~~~~~
Marlin\src\lcd\tft\../menu/menu_item.h: At global scope:
Marlin\src\lcd\tft\../menu/menu_item.h:58:50: error: 'menuAction_t' does not name a type; did you mean 'menu_main'?
     static inline void action(PGM_P const, const menuAction_t func) { (*func)(); };
                                                  ^~~~~~~~~~~~
                                                  menu_main
Marlin\src\lcd\tft\../menu/menu_item.h: In static member function 'static void MenuItem_function::action(const char*, int)':
Marlin\src\lcd\tft\../menu/menu_item.h:58:73: error: invalid type argument of unary '*' (have 'int')
     static inline void action(PGM_P const, const menuAction_t func) { (*func)(); };
                                                                         ^~~~
compilation terminated due to -fmax-errors=5.
Compiling .pio\build\MKS_ROBIN2\src\src\lcd\touch\touch_buttons.cpp.o
*** [.pio\build\MKS_ROBIN2\src\src\lcd\tft\touch.cpp.o] Error 1
Marlin\src\lcd\tft\ui_320x240.cpp:49:4: error: #error "Seriously? High resolution TFT screen without menu?"
   #error "Seriously? High resolution TFT screen without menu?"
    ^~~~~
In file included from Marlin\src\lcd\tft\ui_320x240.cpp:30:0:
Marlin\src\lcd\tft\../menu/menu.h: In static member function 'static void MenuItem_back::action(const char*)':
Marlin\src\lcd\tft\../menu/menu.h:89:63: error: 'class MarlinUI' has no member named 'go_back'
     FORCE_INLINE static void action(PGM_P const=nullptr) { ui.go_back(); }
                                                               ^~~~~~~
Marlin\src\lcd\tft\ui_320x240.cpp: At global scope:
Marlin\src\lcd\tft\ui_320x240.cpp:54:25: error: no 'void MarlinUI::tft_idle()' member function declared in class 'MarlinUI'
 void MarlinUI::tft_idle() {
                         ^
Marlin\src\lcd\tft\ui_320x240.cpp:68:6: error: redefinition of 'static void MarlinUI::init_lcd()'
 void MarlinUI::init_lcd() {
      ^~~~~~~~
In file included from Marlin\src\lcd\tft\ui_320x240.cpp:29:0:
Marlin\src\lcd\tft\../ultralcd.h:328:24: note: 'static void MarlinUI::init_lcd()' previously defined here
     static inline void init_lcd() {}
                        ^~~~~~~~
Marlin\src\lcd\tft\ui_320x240.cpp:78:6: error: redefinition of 'static bool MarlinUI::detected()'
 bool MarlinUI::detected() { return true; }
      ^~~~~~~~
In file included from Marlin\src\lcd\tft\ui_320x240.cpp:29:0:
Marlin\src\lcd\tft\../ultralcd.h:327:24: note: 'static bool MarlinUI::detected()' previously defined here
     static inline bool detected() { return true; }
                        ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\MKS_ROBIN2\src\src\lcd\tft\ui_320x240.cpp.o] Error 1

Rejdok, 2020-10-24:
Unfortinutaly this archive hasn't build with other errors.
I.E.\.platformio\packages\framework-arduinoststm32@4.10700.200103\cores\arduino\stm32\analog.cpp: In function 'uint32_t get_adc_internal_channel(PinName)':
\.platformio\packages\framework-arduinoststm32@4.10700.200103\cores\arduino\stm32\analog.cpp:242:10: error: 'PADC_TEMP' was not declared in this scope
     case PADC_TEMP:
. Actiualy I have much more errors.
But check defenitinons m.b. in configuration file 
Marlin\src\lcd\tft../menu/menu.h:89:63: error: 'class MarlinUI' has no member named 'go_back'
FORCE_INLINE static void action(PGM_P const=nullptr) { ui.go_back(); }
This error caused by undefined "HAS_LCD_MENU". Checkout ultralcd.h .

Rejdok, 2020-10-24:
What is the frequency of the crystal oscillator? Firmware works without display?

LordsWars, 2020-10-24:
Yes, I have the BOARD_BTT_SKR_PRO_V1_1 board installed instead of Robin2 and the pins are all configured for another board ... the boards are the same for the processor ...
8 megahertz
no firmware is not working at all, it is not even defined as an external device

Rejdok, 2020-10-25:
Hm, thaths strange. Robin2 have ovboard usb-uart transmitter, it must works as is and defined as usb cdc.
At first step lets make works firmware without display, then solve other problems.

Rejdok, 2020-10-25:
Heap problems can be solved with changing linker script _Min_Heap_Size = 0x200 to other value. USB stack from stm32 requres a lot of heap memory, and i recomend increase it at least for 16kb.

Rejdok, 2020-10-25:
And one more. Switch usb to fs mode. As I can see this board dosen't have external ULPI HS chip.

Rejdok, 2020-10-25:
Sorry, FS already chosen.

Rejdok, 2020-10-25:
Checkout this code. 
It build without errors. I dosen't change Conf.h and othrer files from main branch of marlin. See pins for mks robin2 I just copy it from btt pins and may be change script for linker.
[Marlin-bugfix-2.0.x.zip](https://github.com/Rejdok/STM32-Marlin-Port/files/5434050/Marlin-bugfix-2.0.x.zip)

Rejdok, 2020-10-25:
Here version with increaced heap/stack size. If I make all right. =) 
[Marlin-2.0.x.zip](https://github.com/Rejdok/STM32-Marlin-Port/files/5434080/Marlin-2.0.x.zip)

LordsWars, 2020-10-25:
Repetier-Host 2.1.6
printer connection com 8
18:17:11.348 : OpenGL version:4.6.0 NVIDIA 391.357
18:17:11.355 : OpenGL renderer:GeForce GT 640/PCIe/SSE2
18:17:11.355 : Using fast VBOs for rendering is possible
18:17:31.165 : No start signal detected - forcing start
18:17:31.200 : N1 M110*34
18:17:31.200 : N2 M115*36
18:17:31.200 : N3 M105*36
18:17:31.200 : N4 M114*35
18:17:31.200 : N5 M111 S6*98
18:17:31.206 : N6 T0*60
18:17:31.206 : N7 M20*22
18:17:31.206 : N8 M80*19
18:17:31.206 : N9 M105*46
18:17:34.222 : N10 M105*22
no response from the board but com port is available
The screen is still without turning on the backlight ...

Rejdok, 2020-10-25:
Yes, it's okay. At display may be black. I dosn't change lcd pins. When you connect with st-link firmware works? It dosn't fall in HardFault? Try change Serial to 3 or 4.

Rejdok, 2020-10-25:
4 or 5, sorry.

Rejdok, 2020-10-25:
If firmware dosn't fail in HardFault I suppose all problem in configuration.

LordsWars, 2020-10-25:
Yes, everything is normally uploaded to the board normally and after the reset there are no board reactions ...
yeah, sort out the ports ... understood, I'll check it ...

Rejdok, 2020-10-25:
Pio have debug task. Please check it. Normal upload doesn't suggests normal operation=)

Rejdok, 2020-10-25:
And in debug mode you can check global varibles and perepherial registers to figure out problem. I dosn't have this board, so I can't debug this code.

LordsWars, 2020-10-25:
according to the wiring diagram, USART3 is used, I see ... he takes the first 6 steps and everything is in the Halted status - and there are no changes in the register ...

LordsWars, 2020-10-25:
[image]

Rejdok, 2020-10-25:
Halted status it's ok. It requred for step by steb debug. Just click on run, then halt. If firmware dosn't in  HardFault everything ok.

LordsWars, 2020-10-25:
a sample of the same screen when the program runs in step-by-step mode
[image]
no, no, it just hangs in one position after 6 steps and there are no changes ... that is, the registers do not change ...

Rejdok, 2020-10-25:
In PeripheralPins USART3 pins on 4 place in array, and if I undertood correctly this shoud be Serial4 or Serial5.

Rejdok, 2020-10-25:
Stec pointer and program counter changing. Program executes.

Rejdok, 2020-10-25:
And now executes startup code not main function. It's ok.

LordsWars, 2020-10-25:
no, they don't change

Rejdok, 2020-10-25:
Hm. Even after click run/halt?

LordsWars, 2020-10-25:
yes, everything is static

Rejdok, 2020-10-25:
And even after click on read core registers after halt?

LordsWars, 2020-10-25:
yes

Rejdok, 2020-10-25:
That's bad

LordsWars, 2020-10-25:
the first 6 steps - there are changes and that's it, there are no further register changes at all.

Rejdok, 2020-10-25:
Hm, in linker script wrong flash size

Rejdok, 2020-10-25:
and place of startup code

LordsWars, 2020-10-25:
[image]
step1
[image]
step2

Rejdok, 2020-10-25:
change to 0x8000000

Rejdok, 2020-10-25:
and reupload

Rejdok, 2020-10-25:
ORIGIN = 0x8000000

LordsWars, 2020-10-25:
[image]

LordsWars, 2020-10-25:
[image]

Rejdok, 2020-10-25:
[image]

Rejdok, 2020-10-25:
Here

LordsWars, 2020-10-25:
it seems that the problem is not in the download address, but in the start of the program address

Rejdok, 2020-10-25:
Yes, change linker script

Rejdok, 2020-10-25:
You can find path to file on screenshot

Rejdok, 2020-10-25:
Your linker script requred bootloader for normal operation. When change flash origin you move start address of programm.

LordsWars, 2020-10-25:
change this line?
FLASH (rx): ORIGIN = 0x800C000, LENGTH = 512K-48K

Rejdok, 2020-10-25:
FLASH (rx)      : ORIGIN = 0x8000000, LENGTH = 512K 
yes

LordsWars, 2020-10-25:
build :)

Rejdok, 2020-10-25:
good =)

[185 more replies]

closed 2020-10-26 by Rejdok